intel/zephyrjs-ide #181: Home page features displaying

qiuzhong, 2017-08-02:
**Issue description**
There is a tiny typo in home page features and feature items are not listed well.

**Current behavior**
Feature items are not listed well.

**Expected behavior**
Feature items are listed like this:
```
* Plug-and-play WebUSB connectivity
* JavaScript Runtime for Zephyr® OS powered boards
* Instant JAvaScript code execution on target
* IDE with multiple tabs, console shell, and GitHub integration
* Web app that works online and offline
* Integrated documentation and board explorer
```

**Steps to reproduce**
Start the IDE and go to home page.
[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
`v0.19.0-alpha / 18c7815`

siovene, 2017-08-02:
The list lacks list indicators (*) by design, but I will fix the typo. Thanks!

closed 2017-08-02 by siovene

qiuzhong, 2017-08-07:
Verified with [4f773db](https://github.com/01org/zephyrjs-ide/commit/4f773db810effa4748bc9158f2e3adeeddf8445f), typo fixed.

commit 2d0169c6 (2017-08-02): fix(home): typo

Closes #181