intel/zephyrjs-ide #187: The connection status is not rigorous

canxcao, 2017-08-09:
**Issue description**
Status bar in "Ready" status even the connecting is cancelled

**Current behavior**
If users cancel the connecting or connect the device, the status bar always return the "Ready" status

**Expected behavior**
If users cancel the connecting or connect the device, the status bar should return the different status.

**Steps to reproduce**
1. Launch the IDE and go to http://localhost:8000
2. Click the "connect", then cancel to connect.
check the value of status bar.
3. Click the "connect", then select the device to connect.
check the value of status bar.
[image]

**'JS IDE for Zephyr OS' version: v0.20.0-alpha/ 4f773db

xiaoyu3x, 2017-09-21:
This issue is still  reproduced  with commit [c5b35be](https://github.com/01org/zephyrjs-ide/commit/c5b35be17d8cc63de91c5c2da4d28308617bf882)

brianjjones, 2018-04-17:
I don't see this issue with the current version, could you please re-test @xiaoyu3x

xiaoyu3x, 2018-04-18:
hello, @brianjjones 
This issue is still produced with commit [02a08ec](https://github.com/intel/zephyrjs-ide/commit/02a08ec5c0d0868a98da8a9ef2c012a218a058bd)
Picture 1:
No connect board and Web IDE, the value of status bar is "Ready"
[image]

Picture 2:
Connect board and Web IDE, the value of status bar still is "Ready"
[image]