intel/zephyrjs-ide #197: Command list information is incomplete when 'help' command is executed in ashell.

xiaoyu3x, 2017-10-18:
**Issue description**
`acm>help` output information is incomplete.

**Current behavior**
There aren't 'load', 'parse' and 'du' command information when execute 'help' command.
[image]

**Expected behavior**
There are 'load', 'parse' and 'du' command information under command list
[image]

**Steps to reproduce**
1.Launch the IDE and connect to the board device.
2. Check help information in console.

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.21.0-alpha / [2796b09](https://github.com/01org/zephyrjs-ide/commit/2796b09bd1d4e87cdad5d4304d7cb66b70600cfa)

siovene, 2017-10-18:
Where did you get the second image from?

xiaoyu3x, 2017-10-18:
hi, @siovene 

One way, you can see link [Ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md#commands) 
Another way, you can get it by below step:
 1. Flash ashell into 101 borad:
   ```
make ROM=256 ashell
   make dfu
```
2. Reset 101 and by minicom see output:

```
sudo minicom -D /dev/ttyACM0
acm>help
```

siovene, 2017-10-18:
Are you sure perhaps you're not comparing whatever you have installed on your 101 to the docs? Maybe you have an older version on your 101, and you're comparing to the docs.

Or, instead, are you comparing the output of `help` from the IDE to the output of `help  from the minicom *using the same 101 flashed with the same ashell*?

xiaoyu3x, 2017-10-18:
hi, @siovene 
I am sure that I use the same 101 to flash IDE and ashell with commit [1d5a99d](https://github.com/01org/zephyr.js/commit/1d5a99d5e2b8b09602e79cd4c1cd1c260eaf224b), but return different info when using 'help' command.
So , I think this is a issue

xiaoyu3x, 2017-10-20:
Relative bug [#6 ](https://github.com/nagineni/zephyrjs-cli/issues/6)

xiaoyu3x, 2018-04-17:
This issue isn't reproduced with commit 02a08ec, so closed it.

closed 2018-04-17 by xiaoyu3x