intel/zephyrjs-ide #216: Redundant character '<!DOCTYPE html>' appears when running IDE with zephyrjs-ide-nginx container.

xiaoyu3x, 2018-01-02:
**Issue description**
`<!DOCTYPE html>` presents on the board explorer when using zephyrjs-ide-nginx container to start IDE page.

**Current behavior**
Redundant character `<!DOCTYPE html>` appears on the board explorer .

**Expected behavior**
There aren't `<!DOCTYPE html>` appears on the board explorer .

**Steps to reproduce**
1.`docker-compose -f docker-compose.production.yml build`
2.`docker-compose -f docker-compose.production.yml up zephyrjs-ide`
3.`docker-compose -f docker-compose.production.yml up -d zephyrjs-ide-nginx`
4. Access http://localhost:8000/#/editor and select 'UP' board of the board explorer.
[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
`v0.22.0-alpha / [1a0a943]`

brianjjones, 2018-04-17:
I don't see this on the latest build, @xiaoyu3x can you re-test?

xiaoyu3x, 2018-04-18:
hi, @brianjjones 
This issue is still produced with commit [e28b227](https://github.com/intel/zephyrjs-ide/commit/e28b227b63b4bf863fd46828b2a18da77c124387)

[image]