intel/zephyrjs-ide #221: Webpag  'localhost:8000' has no response for a long time when execute 'npm start -- --app baz/bar/foo'

xiaoyu3x, 2018-01-23:
**Issue description**
 'localhost:8000' has no response for a long time when execute 'npm start -- --app baz/bar/foo', but run normally in terminal.

**Current behavior**
 'localhost:8000' has no response for a long time

**Expected behavior**
'localhost:8000' upload normally

**Steps to reproduce**
1.`git clone https://github.com/intel/zephyrjs-ide.git`
2.`cd zephyrjs-ide`
3.`npm install`
4.`npm start -- --app baz` or `npm start -- --app foo` or `npm start -- --app bar`
[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.22.0-alpha / [deb9d90](https://github.com/intel/zephyrjs-ide/commit/deb9d90c4b4a1bff526b4fa1fdb66d85774ea47a)

**Comment**
node:v8.9.0
npm:5.5.1

xiaoyu3x, 2018-02-22:
hi, @brianjjones 

I think the root reason of this issue is command  format `npm start -- --app **`  is wrong. 
Correct command  should be `npm start --app **`.
E.g. `npm start --app baz` or `npm start --app foo` or `npm start --app bar` or `npm start --app`

brianjjones, 2018-04-17:
@xiaoyu3x You appear to be correct, I'll update the README. Thanks.

brianjjones, 2018-04-25:
README has been updated, closing.

closed 2018-04-25 by brianjjones

xiaoyu3x, 2018-05-08:
This issue isn't reproduced with commit [6e87086](https://github.com/intel/zephyrjs-ide/commit/6e87086726a83cdfbbfd16e1ecbd1749c059ea79)