intel/zephyrjs-ide #233: Cannot find module 'zephyrjs-ide/tools/env/staging' when running `npm start -- --env-config staging`

xiaoyu3x, 2018-02-22:
**Issue description**
This issue is from `Currently the ENV_NAMEs are dev, prod, staging` of [README](https://github.com/intel/zephyrjs-ide#environment-configuration).
Commands `npm start -- --env-config dev` and `npm start -- --env-config prod` run successfully, but `npm start -- --env-config staging` throws error.

**Current behavior**
`npm start -- --env-config staging` throws error.

**Expected behavior**
`npm start -- --env-config staging` runs successfully, no error

**Steps to reproduce**
1. `git clone https://github.com/intel/zephyrjs-ide.git`
2. `cd zephyrjs-ide`
3. `npm install`
4. `npm start -- --env-config staging`
[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.22.0-alpha / [bae9e36](https://github.com/intel/zephyrjs-ide/commit/bae9e368af04efcafd88bf0185792444ea317ae1)

**Comment**
There isn't file 'staging.ts' leads to happen error under 'zephyrjs-ide/tools/env/'.

xiaoyu3x, 2018-02-22:
hi
I just submitted PR #234 to fix this issue.
This pr has been verified on local successfully and no error happens when running it

brianjjones, 2018-04-17:
This PR has been merged, thanks.

closed 2018-04-17 by brianjjones

xiaoyu3x, 2018-05-08:
This issue isn't reproduced with commit [6e87086](https://github.com/intel/zephyrjs-ide/commit/6e87086726a83cdfbbfd16e1ecbd1749c059ea79)