intel/zephyrjs-ide #235: Fail to save the file into A101

xiaoyu3x, 2018-02-22:
**Issue description**
Can't save any file into A101 when click 'Save' button on WebIDE.

**Current behavior**
'save: command not found' is popups after click 'Save' button

**Expected behavior**
The file is saved successfully and no error throws.

**Steps to reproduce**
1. Build images with commit [ed85a4d](https://github.com/intel/zephyr.js/commit/ed85a4deec7fdc4e6c464ba9bef7c03da3339550) and flash it  into A101 boards.
2. Start WebIDE and connect boards. 
3. Edit any file on editor window and press 'Save' button.
[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.22.0-alpha / [bae9e36](https://github.com/intel/zephyrjs-ide/commit/bae9e368af04efcafd88bf0185792444ea317ae1)

xiaoyu3x, 2018-04-17:
This issue isn't reproduced with commit 02a08ec, so closed it.

xiaoyu3x, 2018-04-17:
Sorry,
This issue still is reproduce on commit [02a08ec](https://github.com/intel/zephyrjs-ide/commit/02a08ec5c0d0868a98da8a9ef2c012a218a058bd)

brianjjones, 2018-04-17:
Using the latest master ZJS and IDE 0.23.0-alpha I don't see this issue. If you see "acm>" in your console window something isn't set up properly.  Make sure and use 'make pristine' before building ZJS to ensure you have the latest and greatest.  If it still fails for you let me know if it fails every time, or only after doing some other things.

Thanks.

xiaoyu3x, 2018-04-18:
hi, @brianjjones 
Picture One:
There isn't file 'b' in board, Device of sidebar is 0
[image]

Picture Two:
Save file 'b' in board, but Device of sidebar still is 0. 
[image]

I use zjs commit [82662d6](https://github.com/intel/zephyr.js/commit/82662d65bb71d3088db9368d390872f9ccf82ede) 

File 'b' hasn't been saved into A101 when using other way to check.

xiaoyu3x, 2018-04-19:
Sorry,
This issue indeed isn't reproduce on commit 02a08ec.
About another save issue, I will renew to file a bug, so closed it.

closed 2018-04-19 by xiaoyu3x