intel/zephyrjs-ide #236: Fail to run code on WebIDE

xiaoyu3x, 2018-02-22:
**Issue description**
Can't run any file when click 'Run' button on WebIDE.

**Current behavior**
'run: command not found' is popups after click 'Run' button

**Expected behavior**
The file is run successfully and no error throws.

**Steps to reproduce**
1. Build images with commit [ed85a4d](https://github.com/intel/zephyr.js/commit/ed85a4deec7fdc4e6c464ba9bef7c03da3339550) and flash it  into A101 board.
2. Start WebIDE and connect board. 
3. Edit any file on editor window and press 'Run' button.
[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.22.0-alpha / [bae9e36](https://github.com/intel/zephyrjs-ide/commit/bae9e368af04efcafd88bf0185792444ea317ae1)

xiaoyu3x, 2018-04-17:
This issue isn't reproduced with commit 02a08ec, so closed it.

closed 2018-04-17 by xiaoyu3x

brianjjones, 2018-04-17:
It works for me using the latest ZJS and IDE v0.23-0-alpha.  @xiaoyu3x could you re-test please?