intel/zephyrjs-ide #242: Redundant character ' } ' appears when saving file into board

xiaoyu3x, 2018-04-17:
**Issue description**
Redundant character ' } '  is printed on console windows after saving file into A101 board 

**Current behavior**
Character  ' } '  appears after saving file into A101 board.

**Expected behavior**
No character  ' } '  appears.

**Steps to reproduce**
1. Build and flash images into board.
2. Start WebIDE and connect board. 
3. Edit any file and click "Save" button to save file into board.
[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.23.0-alpha / [02a08ec](https://github.com/intel/zephyrjs-ide/commit/02a08ec5c0d0868a98da8a9ef2c012a218a058bd)

xiaoyu3x, 2018-04-18:
hello, @brianjjones 
I just created "a" file and save it into A101.
The following is result and redundant character ' } ' appears on console window.
[image]

I use zephyrjs commit [82662d6](https://github.com/intel/zephyr.js/commit/82662d65bb71d3088db9368d390872f9ccf82ede)

xiaoyu3x, 2018-04-19:
Sorry, @brianjjones 
According to your comment, I retested this issue again and found it that happened only when file is without the suffix.
I just updated this issue.
Thank you.

brianjjones, 2018-04-26:
@xiaoyu3x This should be fixed in master now with PR #252

xiaoyu3x, 2018-05-08:
This issue isn't reproduced with commit [6e87086](https://github.com/intel/zephyrjs-ide/commit/6e87086726a83cdfbbfd16e1ecbd1749c059ea79)

closed 2018-05-08 by xiaoyu3x