intel/zephyrjs-ide #243: File "Accelerometer" and "Gyroscope" need to update

xiaoyu3x, 2018-04-17:
**Issue description**
In Examples of sidebar, "Accelerometer" and "Gyroscope" code need to update.

**Current behavior**
Code "sensor.onchange" has been discarded in zephyr.js , but it still be used in WebIDE. 

**Expected behavior**
"Accelerometer" and "Gyroscope" code need to update.

**Steps to reproduce**
Start WebIDE and check  "Accelerometer" and "Gyroscope" in Examples.

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.23.0-alpha / [02a08ec](https://github.com/intel/zephyrjs-ide/commit/02a08ec5c0d0868a98da8a9ef2c012a218a058bd)

brianjjones, 2018-04-17:
Thanks, I've fixed this in the master branch - but it's not live on the webpage yet.  Once it is, I'll reply here that its ready to be re-tested.

brianjjones, 2018-04-26:
Fix is now in master.

closed 2018-04-26 by brianjjones

xiaoyu3x, 2018-05-08:
This issue isn't reproduced with commit [6e87086](https://github.com/intel/zephyrjs-ide/commit/6e87086726a83cdfbbfd16e1ecbd1749c059ea79)