intel/zephyrjs-ide #246: WebUSB device can't be found on windows platform

xiaoyu3x, 2018-04-18:
**Issue description**
WebUSB device can't be found on windows platform

**Current behavior**
Board can't be found after connected windows PC.

**Expected behavior**
Device can be found.

**Steps to reproduce**
1. Build and flash images into board.
2. Start WebIDE and connect board. 
[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.23.0-alpha / [02a08ec](https://github.com/intel/zephyrjs-ide/commit/02a08ec5c0d0868a98da8a9ef2c012a218a058bd)

brianjjones, 2018-04-18:
@nagineni Can you help with this? I know you were the one who set this up last time. I've taken a stab at it by adding back in the windows stuff lost when we switched to using the new webusb driver https://github.com/brianjjones/zephyr.js-1/tree/fix_win . However it still doesn't work, any insight is appreciated.

brianjjones, 2018-04-18:
Also this is a ZJS issue, so I'm opening a bug there https://github.com/intel/zephyr.js/issues/1872

I'll keep this one open until it gets resolved.